brewpi-remix/brewpi-firmware-rmx #30: I2C version of the Arduino crashes

jankeesv, 2021-06-20:
I want to report a hardware related issue I'm facing with my Brewpi Remix setup. At a certain moment after a random number of hours my Arduino UNO freezes when the relais for cooling switches off (see picture).

Hardware setup:
- Raspberry Pi 3
- Arduino UNO R3 (unofficial)
- Relais module powered by Arduino
- I2C and LCD with backlight powered by Arduino
- Arduino is powered by Raspberry Pi USB
- Power supply of Raspberry Pi is a 5V 2A unit

Thoughts so far and possible root causes:
1. 500mA provided by Raspberry Pi is too weak to power Arduino, I2C and Relais module
2. Electromagnetic interference because of the fridge turning off
3. Heat building up in the housing by the Raspberry Pi
 
Did someone have this issue before or seen similar issues? My first thing to try is buy a high graded power supply 5V 6A and power the relais module and I2C directly from the power supply to keep the 500mA bounary of the Arduino safe.
Fallback is to write a script to reboot the Raspberry Pi when Arduino doesn't respond in JSON.

Any help is welcome!

[image]

jankeesv, 2021-06-20:
Yes that was also a thought of me but unfortunately. Quoted: The USB supply to the Arduino is protected by a “Polyfuse” as I described here, which is rated at 550 mA. This means that at currents above this, the Polyfuse will eventually open, and how long it takes to do so is inversely proportional to the degree to which you exceed this current limit (as it is a thermal fuse).

jankeesv, 2021-06-20:
I will power the relais module directly from a stable power source of 5v. The same that will power the pi.

lbussy, 2021-06-20:
I've run one of these for quite a while with no issues, although I do allow the screen to blank.

jankeesv, 2021-06-20:
Did you house the Arduino close to the relais module and 230v wiring? Can be tricky to find with electrical interference.

jankeesv, 2021-06-20:
Remarkable is that the LCD hangs with the text "Idoling" which is a mix of Cooling and Idling. Pretty funny word too. :smile:

lbussy, 2021-06-20:
My relays are in the same box as the Arduino.

Did you use my compiled firmware or did you compile it yourself?

jankeesv, 2021-06-20:
I used the install script when installing the pi.

lbussy, 2021-06-20:
How long ago?  I'm wondering if a recent core upgrade may have impacted things.

If you are using 0.2.13 you might try 0.2.12 and see if there's a difference.  They should be exactly the same for the I2C variant, except the .12 is on an older core.

jankeesv, 2021-06-21:
Ok I will try to downgrade. Is there an easy way to do this with the Python script? Yesterday evening it frozen again and it seems to be a pattern. It's always after the 6th cycle of cooling. And the word "Idoling" came up again.

jankeesv, 2021-06-21:
I've installed my Brewpi some weeks ago for the first time. I was running 0.2.12 already on the Arduino. Right now I've upgraded to the latest version 0.2.13. Keep you updated on this. I'm wondering if this is a memory error due to String operations in the display code?

jankeesv, 2021-06-21:
Same error with the 0.2.13 firmware as the 0.2.12 firmware. Controller freeze and unresponsive after cooling cycle. Text on the display on the last line is also "Idoling". Here the moment it crashes (left).
[image]

lbussy, 2021-06-21:
That's very strange.  And it's always the 6th time?

I have not messed with the strings, the Uno is packed to the gills and I wanted to avoid any issues like this.  WHen I crammed in I2C, I had to remove random letters from strings to fit it in.  I can always see if there's room to save more room but I'd like to reproduce it here first.

Were you able to test external power?

jankeesv, 2021-06-21:
Thank you @lbussy for the close collaboration. I receive a quality 5v supply later this week. Then I can test the separate power of the relais module. If no progress I will replace the Arduino. Let me get back to you on this.

lbussy, 2021-06-21:
I need to finish some of the re-wiring I am doing and I can re-test on a fridge as well.

jankeesv, 2021-06-22:
Couldn't leave it yesterday. I've used the 5V supply to power the relais and I2C board. Result was that it is getting worse. Every disconnect of the fridge led to a frozen Arduino. Now I found this article below to hook up the relais module in a proper way. I'm going to try this tonight to see if the problem solves. Seems like a EMI problem when disconnecting the fridge. https://forum.arduino.cc/t/5v-relay-almost-switching-despite-sufficient-power/424212/7

jankeesv, 2021-06-22:
Had an hour break and hooked up the relais module in this way like the picture. I hope it will fix the problem. 
[image]

lbussy, 2021-06-22:
Is it a very large refrigerator?  While I have used this method, I've not found it necessary in practice.

jankeesv, 2021-06-22:
No its not that large, I'm using a table model. Thing is that a compressor can really cause some spikes in the Arduino circuit. The safe way is to remove the jumper and use the power supply to power that directly. In this way its really seperated and safer. Results are good for the first cycles but I want to see a stable behaviour over a week to move from water to beer ;-)

[image]

jankeesv, 2021-06-22:
Unfortunately another Arduino didn't solve the problem. Also the separate power to the relais module didn't solve it. I now ordered two Solid State Relais that can be powered by the Arduino directly. It's a less error prone solution to limit interference. Have to wait some weeks for them to arrive.

lbussy, 2021-06-22:
I appreciate you reporting back.  On the one hand, I hope it's electronic noise because that will mean it's not something I need to fix.  On the other hand, it should not need an SSR.

If I recall: European power is 230VAC 50Hz?  I wonder if that's somehow inherently noisier?  If so, three MOVs would help smooth out any really big spikes.

jankeesv, 2021-06-23:
Youre welcome. Yes the power is 230VAC 50Hz. The parts are designed for the specifications. Read a lot about the causes these days sobI learn a lot. The issue is because of the ARC effect when releasing the relay of the fridge. This spark causes An electronic interference. It has more todo with the specific fridge than the circuits of brewpi. MOV is a possible solution or a so called "slubber network". I wait for the SSR now. Love these things, I also use Them in my HERMS heaters.

jankeesv, 2021-06-23:
Now we are that far I couldn't resist to buy a simple snubber network. I will receive it later this week. It has two wires and will be placed over the COM and NO contacts of the relay for the fridge. This will limit the negative effects of the ARC.

lbussy, 2021-06-23:
At least in the US, a UL-Listed surge suppressor stip should provide enough transient voltage suppression to make a difference.   It might be easier to find (or you probably have one now).

jankeesv, 2021-06-25:
Solution: use a snubber network of 1uF that you place over the NO and COM of the fridge relay.

[image]

closed 2021-06-25 by jankeesv

lbussy, 2021-06-25:
Glad to hear that worked for you!

jankeesv, 2021-06-25:
Thanx for the collaboration!